francoiscampbell/CLionArduinoPlugin #5: Unable to build Arduino project, missing Wire.h or Time.h when building DS1307RTC library

kigster, 2015-08-13:
Hi! 

I've been playing with this plugin all night yesterday, and while overall I love it and it seems promising, after three hours I could not get it to work, and to successfully build my project.

My `CMakeLists.txt` is basically a template generated one, with my libraries folder added in:
```cmake
cmake_minimum_required(VERSION 3.2)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(PROJECT_NAME Wallock)
project(${PROJECT_NAME})
link_directories(/Users/kig/arduino/libraries)
set(${CMAKE_PROJECT_NAME}_SKETCH Wallock.ino)
generate_arduino_firmware(${CMAKE_PROJECT_NAME})
```

When I run this, get the same error every time, regardless of order of includes, or anything else. The time chip library `DS1307RTC`, when compiling, is unable to find either `Wire.h` or `Time.h`. Note below that both of those libs have been compiled and linked by the time this error is thrown.  This makes me think – could this be just that CLion is trying to build too many things at once, and by the time it gets to this library, something is missing? My other libraries for this project also depend on Wire and Time, but I have not see a single issue with them. Nothing about this library jumps at me – it includes standard `#include <Wire.h>` in the .cpp file, and `#include <Time.h>` in it's header.  And yet, even inside CLion I am unable to click through to these files, even through from other library sources, I can! So for some reason even CLion is somehow differentiating these includes, from the identical includes in other libs.  I am scratching my head!

Here is the build output:

```
/Applications/CLion.app/Contents/bin/cmake/bin/cmake --build /Users/kig/Library/Caches/clion10/cmake/generated/e4107188/e4107188/Debug --target all -- -j 8
Scanning dependencies of target uno_CORE
[  2%] [  7%] [  7%] [ 10%] [ 12%] [ 15%] [ 17%] [ 20%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/abi.cpp.obj
Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp.obj
Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/CDC.cpp.obj
Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial2.cpp.obj
Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial1.cpp.obj
Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp.obj
Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial3.cpp.obj
Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HID.cpp.obj
[ 23%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/IPAddress.cpp.obj
[ 25%] [ 28%] [ 30%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/main.cpp.obj
Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/new.cpp.obj
Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Print.cpp.obj
[ 33%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Stream.cpp.obj
[ 35%] [ 38%] [ 41%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Tone.cpp.obj
Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/USBCore.cpp.obj
Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WMath.cpp.obj
[ 43%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WString.cpp.obj
[ 46%] Building C object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/hooks.c.obj
[ 48%] [ 51%] Building C object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c.obj
[ 53%] Building C object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring.c.obj
[ 56%] Building C object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_analog.c.obj
Building C object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_digital.c.obj
[ 58%] Building C object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_pulse.c.obj
[ 61%] Building C object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_shift.c.obj
Linking CXX static library libuno_CORE.a
[ 61%] Built target uno_CORE
Scanning dependencies of target uno_OneButton
Scanning dependencies of target uno_Encoder
Scanning dependencies of target uno_SimpleTimer
Scanning dependencies of target uno_Time
Scanning dependencies of target uno_Wire
Scanning dependencies of target uno_Adafruit_GFX
[ 71%] [ 71%] [ 71%] [ 71%] [ 74%] [ 79%] [ 82%] [ 82%] Building C object CMakeFiles/uno_Wire.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/utility/twi.c.obj
Building CXX object CMakeFiles/uno_SimpleTimer.dir/Users/kig/arduino/libraries/SimpleTimer/SimpleTimer.cpp.obj
Building CXX object CMakeFiles/uno_Time.dir/Users/kig/arduino/libraries/Time/DateStrings.cpp.obj
Building C object CMakeFiles/uno_Adafruit_GFX.dir/Users/kig/arduino/libraries/Adafruit_GFX/glcdfont.c.obj
Building CXX object CMakeFiles/uno_Encoder.dir/Users/kig/arduino/libraries/Encoder/Encoder.cpp.obj
Building CXX object CMakeFiles/uno_OneButton.dir/Users/kig/arduino/libraries/OneButton/OneButton.cpp.obj
Building CXX object CMakeFiles/uno_Adafruit_GFX.dir/Users/kig/arduino/libraries/Adafruit_GFX/Adafruit_GFX.cpp.obj
Building CXX object CMakeFiles/uno_Wire.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/Wire.cpp.obj
[ 84%] Building CXX object CMakeFiles/uno_Time.dir/Users/kig/arduino/libraries/Time/Time.cpp.obj
Linking CXX static library libuno_Encoder.a
[ 84%] Built target uno_Encoder
Linking CXX static library libuno_OneButton.a
Linking CXX static library libuno_Wire.a
[ 84%] Linking CXX static library libuno_SimpleTimer.a
Built target uno_OneButton
Scanning dependencies of target uno_RotaryEncoderWithButton
[ 84%] Built target uno_Wire
[ 87%] [ 87%] Built target uno_SimpleTimer
Building CXX object CMakeFiles/uno_RotaryEncoderWithButton.dir/Users/kig/arduino/libraries/RotaryEncoderWithButton/RotaryEncoderWithButton.cpp.obj
Linking CXX static library libuno_Time.a
Linking CXX static library libuno_RotaryEncoderWithButton.a
[ 87%] Built target uno_Time
Scanning dependencies of target uno_DS1307RTC
[ 87%] Built target uno_RotaryEncoderWithButton
[ 89%] Building CXX object CMakeFiles/uno_DS1307RTC.dir/Users/kig/arduino/libraries/DS1307RTC/DS1307RTC.cpp.obj
/Users/kig/arduino/libraries/DS1307RTC/DS1307RTC.cpp:26:18: fatal error: Wire.h: No such file or directory
 #include <Wire.h>
                  ^
compilation terminated.
make[2]: *** [CMakeFiles/uno_DS1307RTC.dir/Users/kig/arduino/libraries/DS1307RTC/DS1307RTC.cpp.obj] Error 1
make[1]: *** [CMakeFiles/uno_DS1307RTC.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Linking CXX static library libuno_Adafruit_GFX.a
[ 89%] Built target uno_Adafruit_GFX
make: *** [all] Error 2
```

francoiscampbell, 2015-10-11:
I'm not familiar with that library but I've gotten it to find Wire.h by adding

    link_directories(/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/)

and for Time.h maybe this can help: http://stafava.blogspot.ca/2011/12/arduino-ide-cant-find-my-timeh-file.html

ad-on-is, 2015-10-31:
Had the same issue. 
The solution
1. Update Arduino SDK to 1.6.5
2a. If you get errors that some sources are missing open Terminal and navigate to `/Applications/Arduino.app/Contents/Resources` . If not you're lucky!
2b. Type `ln -s ../Java/ Java``
3. Done

kabroxiko, 2020-09-29:
Go to the local arduino "libraries" folder and remove the hyphen (-). eg.

OneWire-2.3.5 -> OneWire