francoiscampbell/CLionArduinoPlugin #6: Unsure how and where to define my board parameters, and how to upload?

kigster, 2015-08-13:
I am just starting with CMake, and it's unclear to me how much I should be looking at the CMakeArduino project versus this one.

But I am unclear on where and how I define the board type (ie, 'Arduino Mini or Pro Mini'), it's CPU speed ('16Mhz'), the chip ('ATmega328'), voltage (5V) and the port for uploading ('/dev/tty.usbserial-AH02MB22')?

Even having those defined, how do I upload the compiled `*.hex` file using CLion?

Perhaps having that in the README.md would help a lot of noobs like myself, just starting with CMake, but very familiar with C++ and/or Arduino programming.

Thanks!

francoiscampbell, 2015-08-16:
I'll update the readme soon. I'm working on a GUI to choose board parameters. The real goal of this plugin is to abstract the details of CMake away but keep the power of CLion vs the Arduino IDE.

kigster, 2015-08-17:
That sounds wonderful!  If you think the GUI can be ready within a couple days, then you are totally right in that README will be updated differently.  But if it's a bit farther away, please send me a list of very short instructions and I can update README for you. You can reach me at my github username @gmail.com.   Thanks!

francoiscampbell, 2015-10-11:
Hey, sorry for the super late reply, I haven't been able to give any attention to this project (or Arduino at all for that matter) in the last few months. Yes, you can upload from the IDE, you just need to specify the board type and the serial port. 

For example:
set(${CMAKE_PROJECT_NAME}_BOARD uno)
set(${CMAKE_PROJECT_NAME}_PORT /dev/ttys0)

This will create the 'upload' configuration in your Run Configurations and you can click the Run button or use the keyboard shortcut to upload.

I don't see having a lot of time in the future, so I don't think the GUI will be coming along any time soon. You're more than welcome to fork and contribute.

closed 2015-10-11 by francoiscampbell

pedrosmmc, 2019-03-03:
Hi! I use CLion on Windows 10.
How can I address serial com port like "set(${CMAKE_PROJECT_NAME}_PORT /dev/ttys0)" ?
With Arduino support plugin I can connect to COM port, and with avrdude as external tool from command line I can so it as well. But on run (shift + f10) and the CmakeLists configured as:

set(${CMAKE_PROJECT_NAME}_BOARD uno)
set(${CMAKE_PROJECT_NAME}_PORT COM3)
set(mega.build.mcu m16u2)
set(mega.upload.protocol wiring)
set(mega.upload.speed 115200)

The result is:

avrdude: ser_open(): can't open device "COM3": No such file or directory
or
avrdude: ser_open(): can't open device "com3": No such file or directory

Thanks for some help!