francoiscampbell/CLionArduinoPlugin #7: Please update README with simple steps to get existing Arduino project imported

kigster, 2015-08-13:
A lot of things were new and not obvious to me, coming from C++/Arduino background, and Eclipse IDE with Arduino plugin (which I still use and love).

I would imagine most people trying this will have existing projects. These projects will depend on libraries, board types, and require compiling, resolving issues, and uploading to the board. 

It would be amazing to have simple instructions on how to do that.

Thanks!

francoiscampbell, 2015-08-16:
Project import via the GUI is on the way. If you have 3rd party libs in the libraries folder and the Arduino IDE picks them up, then CMake should too. I'll investigate if that's not the case.

kigster, 2015-08-17:
Gotcha, so I am NOT actually using Arduino IDE.  I install it, and then typically use Eclipse IDE + Eclipse Arduino plugin.

kigster, 2015-08-17:
Would you be interested in any help? Is there an easy way to split the project into non-overlapping features? I am very comfortable with IntelliJ platforms (IDEA since 2005, RubyMine since pre-release, and CLion since pre-release).  I've never written a plugin but I am sure I can pick it up quick.

francoiscampbell, 2015-10-11:
Yeah, I'd love some help. Like I mentioned in #6, I don't have a lot of time to devote to this; I'm a student working full-time on a co-op term. My next goal is to have GUI options to set the firmware and board parameters and not have to go through the CMakeLists.txt file for it.

closed 2015-10-11 by francoiscampbell