francoiscampbell/CLionArduinoPlugin #8: how to install

az-z, 2015-10-12:
thank you for the plugin.

I installed (fedora) Clion and then this plugin. But there is no "Arduino" menu item i can find.
I have 1.6.5 Arduino IDE installed.

francoiscampbell, 2015-10-12:
For now, there is only the option to create a new Arduino project from the Welcome screen or the File menu, which sets up CMake to pull the Arduino SDK. From there, you can follow the documentation at https://github.com/queezythegreat/arduino-cmake to choose your board parameters and upload. The instructions to upload are in this repo's README

closed 2015-10-12 by francoiscampbell

az-z, 2015-10-13:
aha! now i see it. Clion updated the plugin to the latest version today and the "new project" as well as "new file" dialogues have the arduino option.
Thank you!