francoiscampbell/CLionArduinoPlugin #9: Unable to set mega2560 (or mega1280)

ronny332, 2015-10-20:
Hi,

it seems there is a kind of bug if you try to set the board type to a Atmega2560 or Atmega1280.
If've searched the whole evening for the reason, but the problem seems to be located at the cmake parser file, which tries to separate the different board types.

My boards.txt has the type "mega" for a board with Atmega1280 AND Atmega2560. It is impossible to make a real decision for one of the types and at least the mcu values missing afterwards (what results in a missing compiler flag: -mmcu).

The issue can easily be validated, just run print_board_settings(mega) from your CMakeLists.txt. The .mcu value is missing. print_board_settings(uno) shows it in the right way.

jgenender, 2015-11-14:
Yep... I get it too for the nano (atmega328p).  The only fix (after setting the *_BOARD, *_PORT) is the following:

```
set(nano.build.mcu atmega328p)
set(nano.upload.protocol arduino)
set(nano.upload.speed 57600)
```

This is definitely a bug and a hack to work around it.

francoiscampbell, 2016-01-20:
Sorry for the absence, I've been super busy lately. This is an issue with the arduino-cmake project I use in the plugin. That project is super fragmented right now and hasn't been updated in over a year, so I've got my own form now to try to collect all the fixes and maybe get developing going again. Anyways, as a workaround, I use megaADK personally and it works fine for my mega2560.

mucamaca, 2016-05-13:
With this code:
`cmake_minimum_required(VERSION 2.8.4)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(PROJECT_NAME Arduino_C_things)
project(${PROJECT_NAME})
set(${CMAKE_PROJECT_NAME}_BOARD mega)
set(${CMAKE_PROJECT_NAME}_PORT COM8)
set(${CMAKE_PROJECT_NAME}_SKETCH Arduino_C_things.ino)
set(mega.build.mcu atmega2560)
set(mega.upload.protocol wiring)
set(mega.upload.speed 115200)
generate_arduino_firmware(${CMAKE_PROJECT_NAME})`
The code compiles and downloads on original Arduino Mega, but after that, code is not executed on Arduino. Tested it with simple `Serial.println()` and nothing appeared in serial monitor. Also tried flashing the onboard LED with no result.