francoiscampbell/CLionArduinoPlugin #10: "Failure to initialize" after 1.2.0 Update

GItThatSparrow, 2015-10-26:
Updated the plugin and, upon starting CLion, getting the following error message in the dialog:

```java
Plugin 'io.github.francoiscambell.clionarduinoplugin' failed to initialize and will be disabled.  Please restart CLion.
 
com.intellij.diagnostic.PluginException: io/github/francoiscambell/clionarduinoplugin/components/ArduinoFileTypeRegistration : Unsupported major.minor version 52.0 [Plugin: io.github.francoiscambell.clionarduinoplugin]
    at com.intellij.ide.plugins.cl.PluginClassLoader.b(PluginClassLoader.java:130)
    at com.intellij.ide.plugins.cl.PluginClassLoader.a(PluginClassLoader.java:77)
    at com.intellij.ide.plugins.cl.PluginClassLoader.loadClass(PluginClassLoader.java:66)
    at java.lang.ClassLoader.loadClass(ClassLoader.java:247)
    at java.lang.Class.forName0(Native Method)

// skip a bunch of lines and then....

Caused by: java.lang.UnsupportedClassVersionError: io/github/francoiscambell/clionarduinoplugin/components/ArduinoFileTypeRegistration : Unsupported major.minor version 52.0
    at java.lang.ClassLoader.defineClass1(Native Method)
    at java.lang.ClassLoader.defineClassCond(ClassLoader.java:637)
```

Looks like an issue registering the `.ino` file type, perhaps.  Not sure - I'm not familiar with the JetBrains APIs.

francoiscampbell, 2015-10-27:
Oops, seems I built for Java 8 by accident. I've rebuilt in version 1.2.1 which should be available in a day or two, pending approval from JetBrains.

closed 2015-10-27 by francoiscampbell

GItThatSparrow, 2015-11-14:
Awesome - thanks!  Funny, I knew you updated it since, within a day or two, CLion notified me of the update.  Well done - I appreciate the effort!