francoiscampbell/CLionArduinoPlugin #11: How to set ARDUINO_SDK_PATH?

hameno, 2015-11-10:
Hi,
I installed the plugin and created a new Sketch project, but it just does not seem to work. Cmake shows, that it cannot find "ARDUINO_SDK_PATH", also "CMAKE_C_COMPILER_ENV_VAR" and "CMAKE_CXX_COMPILER_ENV_VAR" are missing. 
I'm using the plugin with CLion 1.2.

SergeySn, 2015-11-10:
I have the same issue:
**Error:Could not find Arduino SDK (set ARDUINO_SDK_PATH)!**
I'm using the plugin 1.2.1 with CLion 1.1.1 on Linux Mint with Arduino SDK 1.6.

madhead, 2015-11-10:
Having the same issue. I'm using CLion 1.2 / CLion Arduino 1.2.1 / Arduino 1.6.6 installed at `/tools/arduino`. `ARDUINO_SDK_PATH` is set:

```shell
┌[madhead@madhead-linux:/proc/14379]
└─$ cat "/proc/$(ps axu | grep -v grep | grep 'com.intellij.idea.Main' | awk '{print $2}')/environ" | awk -v 'RS=\0' -F= '$1=="ARDUINO_SDK_PATH" {print $2}'           
/tools/arduino
```

dmitry-cherkas, 2015-12-01:
To resolve this issue, make sure to put Arduino SDK under one of search paths:
{code}
    if(UNIX)
        file(GLOB SDK_PATH_HINTS /usr/share/arduino*
            /opt/local/arduino*
            /opt/arduino*
            /usr/local/share/arduino*)
    elseif(WIN32)
        set(SDK_PATH_HINTS "C:\\Program Files\\Arduino"
            "C:\\Program Files (x86)\\Arduino"
            )
    endif()
{code}

I believe, you can also alter search paths locally, it is located at cmake/ArduinoToolchain.cmake:55 inside your generated project.

fkrauthan, 2015-12-19:
I have that issue too and my arduino is installed at "C:\\Program Files (x86)\\Arduino"

fkrauthan, 2015-12-19:
Ok I am one step further if using Cygwin on Windows with clion I had to modify the SDK_PATH_HINTS for UNIX. So my version looks like this:

```
    if(UNIX)
        file(GLOB SDK_PATH_HINTS /usr/share/arduino*
            /opt/local/arduino*
            /opt/arduino*
            /usr/local/share/arduino*
            /cygdrive/c/Program\ Files\ \(x86\)/Arduino)
    elseif(WIN32)
        set(SDK_PATH_HINTS "C:\\Program Files\\Arduino"
            "C:\\Program Files (x86)\\Arduino"
            )
    endif()
```

Note the new line `/cygdrive/c/Program\ Files\ \(x86\)/Arduino`. For completion we probably should include `/cygdrive/c/Program\ Files/Arduino`

But now I get `Error:The C compiler "/cygdrive/c/Program Files (x86)/Arduino/hardware/tools/avr/bin/avr-gcc.exe" is not able to compile a simple test program.` which happens because of 

```
avr-gcc.exe: error: /cygdrive/c/Users/fkrauthan/.CLion12/system/cmake/generated/9a83108/9a83108/__default__/CMakeFiles/CMakeTmp/testCCompiler.c: No such file or directory
```

So I guess we can't use Cygwin and this plugin or do you guys have any ideas?

dmitry-cherkas, 2015-12-19:
AFAIK, this does not work with cygwin on Windows. Consider switching to
mingw.

fkrauthan, 2015-12-20:
Yeah tried it with mingw and everything is working fine. Should be mentioned in the plugin description.

efodor, 2015-12-31:
Had similar problems with cygwin, but mingw seemed to resolve similar issues I was running into with cygwin.

r3bers, 2016-04-02:
In Windows 10 I have ARDUINO folder on my Desktop, for support ArduinoToolchain.cmake I created symlink from «\Program Files\Arduino» folder to «\Users\user\Desktop\ARDUINO» folder. But compiler don't see «libraries» that located in «\Users\user\Document» folder. I use This strange folder structure because libriary updates and new library installations from Arduino IDE repository do it in this structure, it helps change version of IDE and SDK by deleting old ARDUINO folder from Desktop and unzip new without library reinstalling.

And I have a question. How change cmake file to look for librares in that way?

r3bers, 2016-04-02:
A find directive for CMake
link_directories(/Users/user/Documents/Arduino/libraries)
But I think its rude to do it every time by hands.
Can Cmake get Shell variables like %HOMEPATH% from Windows env?

r3bers, 2016-04-02:
Some research and: link_directories($ENV{HOMEPATH}/Documents/Arduino/libraries)
And don't forget what some Arduino libraries dirs don't have dirname same as .h file you need, and Cmake don't like it.

twanek, 2016-07-05:
hello!

i have the same problem, using windows 10, clion 2016.1, arduino ide 1.6.9
the arduino libraries are in c:\Program Files (x86)\Arduino\libraries\

i do not know how to set up clion to use the arduino libs. at compilation, 
[image]
got:
==============================================
C:/Users/wanek/ClionProjects/UNO/UNO.ino:53:63: fatal error: SPI.h: No such file or directory

#include <SPI.h> //for digipot serial interface
^
compilation terminated.
==============================================

clion lists them in the external libraries folder, but they are not active. i permanently got the: 'some source files are located outside of cmakelists.txt directory. you can change the project root, or ignore this in future' warning (you can see in the attached image).

please help!
thank you

pattrickrice, 2017-12-22:
Was this issue resolved? I am experiencing this too Win 10 using Clion

lawrencepn, 2018-08-30:
For those using MAC, this is your SDK Location: 
`set(ARDUINO_SDK_PATH /Applications/Arduino-2.app/Contents/java)`

krischik, 2020-11-01:
To bad that there are no macOS informations.

davelnewton, 2021-06-15:
> `set(ARDUINO_SDK_PATH /Applications/Arduino-2.app/Contents/java)`

Only if that's where your Arduino app is, by default it's `Arduino.app`. It's `.../Java`, although the case-sensitivity may depend on your FS settings.

Setting this makes everything "work", although it brings in all the underlying Arduino lib code into the project view and I haven't figured out how to make it go away yet. It's not clear how to set the port without checking `/dev` and manually pasting it in to the `CMakeLists.txt` file.

ag039st, 2023-09-03:
what if I don't have the path /Applications/Arduino-2.app/Contents/java on my M1 I've until /Applications/Arduino-2.app/Contents/ , but not Java anywhere there