francoiscampbell/CLionArduinoPlugin #13: make[2]: *** No rule to make target `libuno_Wire.a', needed by `ArduinoDevelopment.elf'

IT2BE, 2016-01-16:
Hi,

I just started playing with CLion and your Arduino Plugin.

Looks promising but I get this message.

Can you tell me how I can solve it?

This is my CMakeList.txt file:
--- CMakeList.txt ---
cmake_minimum_required(VERSION 2.8.4)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(PROJECT_NAME ArduinoDevelopment)
project(${PROJECT_NAME})

set(${CMAKE_PROJECT_NAME}_SKETCH ArduinoDevelopment.ino)
generate_arduino_firmware(${CMAKE_PROJECT_NAME})

# Include Libraries
include_directories(~/Documents/ArduinoDevelopment2/libraries/MySensors)
include_directories(~/Documents/ArduinoDevelopment2/libraries/SPIFlash)
include_directories(~/Documents/ArduinoDevelopment2/libraries/DHT)

# Board & Serial Port Settings
set(${CMAKE_PROJECT_NAME}_BOARD uno)
set(${CMAKE_PROJECT_NAME}_PORT /dev/ttys0)

This is the output:
--- Messages---
/Applications/CLion.app/Contents/bin/cmake/bin/cmake --build /Users/marceltrapman/Library/Caches/CLion12/cmake/generated/557cf18e/557cf18e/Debug --target all -- -j 4
Scanning dependencies of target uno_CORE
[  7%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/CDC.cpp.obj
[  7%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/abi.cpp.obj
[ 14%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp.obj
[ 14%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp.obj
[ 17%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial1.cpp.obj
[ 21%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial2.cpp.obj
[ 25%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial3.cpp.obj
[ 28%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/IPAddress.cpp.obj
[ 32%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/main.cpp.obj
[ 35%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/new.cpp.obj
[ 39%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/PluggableUSB.cpp.obj
[ 42%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Print.cpp.obj
[ 46%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Stream.cpp.obj
[ 50%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Tone.cpp.obj
[ 53%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/USBCore.cpp.obj
[ 57%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WMath.cpp.obj
[ 60%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WString.cpp.obj
[ 64%] Building C object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/hooks.c.obj
[ 67%] Building C object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c.obj
[ 71%] Building C object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring.c.obj
[ 75%] Building C object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_analog.c.obj
[ 78%] Building C object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_digital.c.obj
[ 82%] Building C object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_pulse.c.obj
[ 85%] Building C object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_shift.c.obj
[ 89%] Linking CXX static library libuno_CORE.a
[ 89%] Built target uno_CORE
[ 92%] Regnerating ArduinoDevelopment.ino Sketch
-- Generating ArduinoDevelopment
-- Generating uno_Wire for library Wire
-- Generating uno_SPI for library SPI
-- Using /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-objcopy for converting firmware image to hex
-- Configuring done
-- Generating done
-- Build files have been written to: ~/Library/Caches/CLion12/cmake/generated/557cf18e/557cf18e/Debug
Scanning dependencies of target ArduinoDevelopment
make[2]: *** No rule to make target `libuno_Wire.a', needed by `ArduinoDevelopment.elf'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 96%] Building CXX object CMakeFiles/ArduinoDevelopment.dir/ArduinoDevelopment_ArduinoDevelopment.ino.cpp.obj
make[1]: *** [CMakeFiles/ArduinoDevelopment.dir/all] Error 2
make: *** [all] Error 2

ruswerner, 2016-12-08:
Has there been any resolution to this? I'm also having the same problem...

ruswerner, 2016-12-08:
After some messing around (for probably way too long), I got it to work by using `include_directories()` in the `CMakeLists.txt` file. I had only `link_directories()` there and that didn't work on its own. Hopefully this helps some people out. This file should probably be generated with some commented out and documented lines for how to include libraries.