francoiscampbell/CLionArduinoPlugin #14: avr-g++: error: missing argument to '-mmcu='

raulgotor, 2016-01-19:
Hello all,

I'm getting this error when trying to compile a simple project. Does anybody of you got it before?

**LOG**
/Applications/CLion.app/Contents/bin/cmake/bin/cmake --build /Users/raulgotor/Library/Caches/CLion12/cmake/generated/3bc8a824/3bc8a824/Debug --target untitled3-size -- -j 4
[  3%] Building CXX object CMakeFiles/nano_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/abi.cpp.obj
[  6%] Building CXX object CMakeFiles/nano_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/CDC.cpp.obj
[ 10%] Building CXX object CMakeFiles/nano_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp.obj
[ 13%] Building CXX object CMakeFiles/nano_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp.obj
avr-g++: error: missing argument to '-mmcu='
make[3]: *** [CMakeFiles/nano_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/abi.cpp.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
avr-g++: error: missing argument to '-mmcu='
make[3]: *** [CMakeFiles/nano_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/CDC.cpp.obj] Error 1
avr-g++: error: missing argument to '-mmcu='
avr-g++: error: missing argument to '-mmcu='
make[3]: *** [CMakeFiles/nano_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp.obj] Error 1
make[3]: *** [CMakeFiles/nano_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp.obj] Error 1
make[2]: *** [CMakeFiles/nano_CORE.dir/all] Error 2
make[1]: *** [CMakeFiles/untitled3-size.dir/rule] Error 2
make: *** [untitled3-size] Error 2

***My CMake configuration***

cmake_minimum_required(VERSION 2.8.4)

set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(PROJECT_NAME untitled3)
project(${PROJECT_NAME})

set(${CMAKE_PROJECT_NAME}_SKETCH untitled3.ino)
set(${CMAKE_PROJECT_NAME}_BOARD nano)
set(${CMAKE_PROJECT_NAME}_PORT /dev/tty.wchusbserial1a1210)

generate_arduino_firmware(${CMAKE_PROJECT_NAME})

**Many thanks!!**

francoiscampbell, 2016-01-20:
The base arduino-cmake project has been split into a bunch of forks. I've created my own to try to corral the features back together. One of the changes I pulled in affected the setting of this argument. I'm not able to reproduce this issue, so here's a test build for you to try. Install it by going to your CLion plugins and choosing 'Install plugin from disk'.
[CLionArduinoPlugin.zip](https://github.com/francoiscampbell/CLionArduinoPlugin/files/98377/CLionArduinoPlugin.zip)

raulgotor, 2016-01-21:
Hello Francois,

Thank you for your answer, and for developing the plugin.

Unfortunately, with this test build I get the following IDE error when creating a new project (MacOSX 10.11.1, CLion 1.2.2, Java 7):

org/apache/commons/io/IOUtils
java.lang.NoClassDefFoundError: org/apache/commons/io/IOUtils
	at io.github.francoiscambell.clionarduinoplugin.resources.ArduinoToolchainFiles.closeStreams(ArduinoToolchainFiles.java:59)
	at io.github.francoiscambell.clionarduinoplugin.resources.ArduinoToolchainFiles.access$000(ArduinoToolchainFiles.java:15)
	at io.github.francoiscambell.clionarduinoplugin.resources.ArduinoToolchainFiles$1.run(ArduinoToolchainFiles.java:45)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWriteAction(ApplicationImpl.java:1002)
	at io.github.francoiscambell.clionarduinoplugin.resources.ArduinoToolchainFiles.copyToDirectory(ArduinoToolchainFiles.java:25)
	at io.github.francoiscambell.clionarduinoplugin.wizards.NewArduinoProjectWizard.createProject(NewArduinoProjectWizard.java:95)
	at io.github.francoiscambell.clionarduinoplugin.wizards.NewArduinoProjectWizard.tryFinish(NewArduinoProjectWizard.java:63)
	at com.jetbrains.cidr.cpp.cmake.projectWizard.CMakeProjectWizard.doOKAction(CMakeProjectWizard.java:61)
	at com.intellij.ide.wizard.AbstractWizard.access$200(AbstractWizard.java:49)
	at com.intellij.ide.wizard.AbstractWizard$5.actionPerformed(AbstractWizard.java:188)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6535)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6300)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4891)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4713)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2750)
	at java.awt.Component.dispatchEvent(Component.java:4713)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at com.intellij.ide.IdeEventQueue.b(IdeEventQueue.java:866)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:650)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:381)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:109)
	at java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:184)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:229)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:227)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:227)
	at java.awt.Dialog.show(Dialog.java:1084)
	at com.intellij.openapi.ui.impl.DialogWrapperPeerImpl$MyDialog.show(DialogWrapperPeerImpl.java:792)
	at com.intellij.openapi.ui.impl.DialogWrapperPeerImpl.show(DialogWrapperPeerImpl.java:465)
	at com.intellij.openapi.ui.DialogWrapper.invokeShow(DialogWrapper.java:1638)
	at com.intellij.openapi.ui.DialogWrapper.show(DialogWrapper.java:1587)
	at com.jetbrains.cidr.cpp.cmake.projectWizard.CMakeProjectWizard.runWizard(CMakeProjectWizard.java:70)
	at io.github.francoiscambell.clionarduinoplugin.actions.NewArduinoProject.actionPerformed(NewArduinoProject.java:26)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performActionDumbAware(ActionUtil.java:182)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter$1.run(ActionMenuItem.java:311)
	at com.intellij.openapi.wm.impl.FocusManagerImpl.runOnOwnContext(FocusManagerImpl.java:958)
	at com.intellij.openapi.wm.impl.IdeFocusManagerImpl.runOnOwnContext(IdeFocusManagerImpl.java:124)
[... 11 lines cut ...]
	at java.awt.MenuComponent.dispatchEvent(MenuComponent.java:339)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:761)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
[... cut ...]
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:381)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

Best regards,
Raúl

francoiscampbell, 2016-01-21:
That's weird, you've got almost the same setup as me but I don't get that issue. Here's a fresh ZIP that I tested working: 
[CLionArduinoPlugin.zip](https://github.com/francoiscampbell/CLionArduinoPlugin/files/99297/CLionArduinoPlugin.zip)

I'm in the middle of a possible Kotlin re-write which would remove the need for IOUtils anyway.

raulgotor, 2016-01-23:
Hi François,

Thanks for your answer and the file. Still no luck. I found out that if the BOARD is set "uno", compiles like a charm, but I cannot fully test it as I don't have any uno laying around.

On the other hand, with BOARD "nano" I get the previous reported error, and with "mega2560", "diecimilia" or "atmega8", I get the following, which makes me think that may be there is something else to be configured on my setup, as it says that mega2560 is an invalid BOARD_ID.

Thanks in advance,
Raúl

/Applications/CLion.app/Contents/bin/cmake/bin/cmake --build /Users/raulgotor/Library/Caches/CLion12/cmake/generated/88180709/88180709/Debug --target upload -- -j 4
-- Generating ard
CMake Error at cmake/Platform/Arduino.cmake:847 (message):
  Invalid Arduino board ID (mega2560), aborting.
Call Stack (most recent call first):
  cmake/Platform/Arduino.cmake:1104 (get_arduino_flags)
  cmake/Platform/Arduino.cmake:516 (setup_arduino_target)
  CMakeLists.txt:10 (generate_arduino_firmware)

-- Configuring incomplete, errors occurred!
See also "/Users/raulgotor/Library/Caches/CLion12/cmake/generated/88180709/88180709/Debug/CMakeFiles/CMakeOutput.log".
make: *** [cmake_check_build_system] Error 1

ad-on-is, 2016-07-22:
hi,

same here!
when setting board to "nano" i get the `missing argument to '-mmcu='`error.

did somebody find a solution?

argestes, 2016-07-31:
I'm getting same error

pheerai, 2016-10-21:
Just for the protocoll, and that other people can find this:

You get this error whenever the you use a board that has been built with several atmega-chips.

To fix this issue, add
```cmake
set(ARDUINO_CPU=<yourCPUtype>
```
BEFORE setting of `CMAKE_TOOLCHAIN_FILE`

`<yourCPUtype>` usually is of the form `atmegaX`, with X being the number from the chip (in my case with Duemilanove, it's `atmega328`)

zero-system, 2018-03-07:
It works for me when I leave out the "=" sign in `set(ARDUINO_CPU=<yourCPUtype>)`

`
cmake_minimum_required(VERSION 2.8.4)

set(ARDUINO_CPU atmega328) # does not need '=' sign

set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(PROJECT_NAME test)
project(${PROJECT_NAME})

set(${CMAKE_PROJECT_NAME}_SKETCH test.ino)

#### Uncomment below additional settings as needed.
set(${CMAKE_PROJECT_NAME}_BOARD nano)
set(${CMAKE_PROJECT_NAME}_PORT COM3) # COM3 on Windows

# set(mega.build.mcu atmega2560)
# set(mega.upload.protocol wiring)
# set(mega.upload.speed 115200)

generate_arduino_firmware(${CMAKE_PROJECT_NAME})
`