francoiscampbell/CLionArduinoPlugin #16: .ino files are not refactor-able because CLion doesn't think they're part of the project

SuperTango, 2016-01-21:
Looks like although CLion can do some refactoring of .ino/.pde files, it can't do the full complement (e.g. inline or extract variable) because it doesn't think the sketch file is part of the source base.  

[image]

In this image, I'm trying to "extract variable" on "millis()".

BildPeter, 2016-02-09:
I also have problems with my refactoring. Doesn't actually work at all, when I get the prompt, still no refactoring is happening. 
Search&Replace does, but it's more work

[image]

SuperTango, 2016-02-09:
I found a workaround.  It's not optimal but it gets lets CLion do more of what it's good at (refactoring, warnings, etc).  

In the CMakeLists.txt file, add 

```
set(SOURCE_FILES sketch.ino)
add_executable(${PROJECT_NAME}2 ${SOURCE_FILES})
```

Now CLion will show the .ino file as not greyed out in the files list, and the refactorings and such will work. 

[image]

I'm fairly certain that this is NOT the "right" way to do it.  CMake complains about not being able to determine linkages, and it can't actually build.  But I'm only using CLion for writing code (I still use the Arduino IDE for building, uploading, and serial monitor), so this lets more of the code writing functions work.

SuperTango, 2016-02-11:
I found a better way to do it, though still not sure if it's optimal.  

Instead of what I said in the last comment, add a "SRCS" parameter to the generate_arduino_firmware command:

```
generate_arduino_firmware(${CMAKE_PROJECT_NAME}
        SRCS sketch.ino)
```

BildPeter, 2016-02-12:
That's excellent! 
It even works for multiple files, as long as you write them into one row (otherwise only the last will be taken)
```
generate_arduino_firmware(${CMAKE_PROJECT_NAME}
        SRCS sketch.ino math.ino extras.ino)
```

francoiscampbell, 2016-02-12:
Hey guys, thanks for pointing this out. This seems to be an issue with the arduino-cmake toolchain, not the CLion plugin per se. It seems that arduino-cmake development has pretty much stalled, so I've got my own fork and I've tried to implement most of the fixes from previous forks of the project, which are also mostrly dead. This project was just an exercise in making a JetBrains plugin that turned out well, but I don't have a lot of free time on my hands to do this though, what with full-time university and what not, so I can't give a definite date on what will be fixed when, but I will make an effort to get it done.

BildPeter, 2016-02-14:
Thank you very much! I really appreciate it

francoiscampbell, 2016-02-17:
Fixed in 3083553. (arduino-cmake commit francoiscambpell/arduino-cmake@dbd4fe3)

closed 2016-02-17 by francoiscampbell