francoiscampbell/CLionArduinoPlugin #17: How to include headers

BildPeter, 2016-02-04:
> undefined reference to `Servo::Servo()'
`undefined reference to `Servo::Servo()'`
`#include <Servo.h>

void setup() {
    Servo one;
}

void loop() {
}`

francoiscampbell, 2016-02-04:
In your CmakeLists.txt file, add

    include_directories(<sketchbook location>/libraries/Servo)

where `<sketchbook location>` is the full path to your Arduino sketchbook folder.

BildPeter, 2016-02-04:
If I copy now the Servo files from the SDK (mac) into the location of my CLion project, then the header should be found. But the .cpp files are in another folder, e.g.
`${CMAKE_SOURCE_DIR}/Servo/avr`
Anyway, if I copy the files into the same folder, it still doesn't work:
```
project(${PROJECT_NAME})
include_directories( ${CMAKE_SOURCE_DIR}/Servo )
```

Does this mean, each **standard** Arduino Lib has to be copied into another directory?
The Servo Lib is shown in CLion as "External Libraries".

francoiscampbell, 2016-02-04:
Aah, seems the Servo library has a different folder structure than Arduino-cmake expects. add `set(Servo_RECURSE True)` right before `generate_arduino_firmware(${CMAKE_PROJECT_NAME})` and it should work.

As a note, the required folder stucture is

    ExampleLib/
      |-- ExampleLib.h
      |-- ExampleLib.cpp
      `-- OtherLibSource.cpp

but the Servo library has

    Servo/
        |-- src/
            |-- avr
                |-- Servo.cpp
                |-- ServoTimers.h
            |-- sam
                |-- Servo.cpp
                |-- ServoTimers.h
            |-- Servo.h

closed 2016-02-04 by francoiscampbell

BildPeter, 2016-02-04:
That solved it! Merci!