francoiscampbell/CLionArduinoPlugin #22: OSX - Error:Could not find Arduino SDK (set ARDUINO_SDK_PATH)!

carlskii, 2016-07-27:
I get this error. Does this plugin work? There seems to be no documentation on how to configure it. I've downloaded and can use the Arduino (1.6) IDE but how do I reference it from the plugin.

Tayphoon, 2016-11-12:
You could set it like this:  set(ARDUINO_SDK_PATH /User/YOUR_ARDUINO_PATH)

You could read about how to use it here https://github.com/francoiscampbell/arduino-cmake

krischik, 2020-11-01:
`/User/YOUR_ARDUINO_PATH`? are you sure? Not `/Application/Arduino.app/…`?

Anyway, how can I spot the correct path? What kind of files or directory would be in there? If I knew one file which should be at the location I can usually figure it out.