francoiscampbell/CLionArduinoPlugin #29: [Issue] 'New Arduino Sketch Project' does not seem to work after upgrading to CLion 3.2

mcfongtw, 2016-12-29:
Hello,

After upgrading to CLion 3.2, the 'New Arduino Sketch Project' on welcome page does not work, where as in CLion 3.1 before upgrading, there should be a pop-up window after clicking it.

And a ClassNotFoundException was thrown when opening a non-arduino project (regular cpp project) using the same IDE.

java.lang.NoClassDefFoundError: com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
	at io.github.francoiscambell.clionarduinoplugin.actions.NewArduinoProject.actionPerformed(NewArduinoProject.java:26)
	at com.intellij.ui.components.labels.ActionLink$1.linkSelected(ActionLink.java:66)
	at com.intellij.ui.components.labels.LinkLabel.doClick(LinkLabel.java:156)
	at com.intellij.ui.components.labels.ActionLink.doClick(ActionLink.java:79)
	at com.intellij.ui.components.labels.LinkLabel$MyMouseHandler.mouseReleased(LinkLabel.java:341)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at com.intellij.ide.IdeEventQueue.i(IdeEventQueue.java:843)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:675)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:391)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.ClassNotFoundException: com.jetbrains.cidr.cpp.cmake.projectWizard.NewCMakeProjectStepAdapter PluginClassLoader[io.github.francoiscambell.clionarduinoplugin, 1.2.2]
	at com.intellij.ide.plugins.cl.PluginClassLoader.loadClass(PluginClassLoader.java:64)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 39 more

mcfongtw, 2016-12-29:
I forgot to mention that neither the 'New Arduino Sketch Project' under 'File' drop down menu would work.

Nihlus, 2017-01-02:
I've got the same issue on Mint 18.

CLion 2016.3.2
Build #CL-163.10154.43, built on December 21, 2016
Licensed to Jarl Gullberg
Subscription is active until August 2, 2017
JRE: 1.8.0_112-release-408-b6 amd64
JVM: OpenJDK 64-Bit Server VM by JetBrains s.r.o

Exception:
https://gist.github.com/Nihlus/379848e22ade850a8bbd6548411f92a7

frozenjava, 2017-01-06:
I am experiencing the same problem in Clion 3.2 on OSX.

ltearno, 2017-01-06:
I have the same problem when doing new Arduino Sketch project.
On Windows 10 64 bits.

Here is the stack trace

com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
java.lang.NoClassDefFoundError: com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
	at io.github.francoiscambell.clionarduinoplugin.actions.NewArduinoProject.actionPerformed(NewArduinoProject.java:26)
	at com.intellij.openapi.actionSystem.ex.ActionUtil$1.run(ActionUtil.java:199)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performActionDumbAware(ActionUtil.java:215)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.a(ActionMenuItem.java:315)
	at com.intellij.openapi.wm.impl.FocusManagerImpl.runOnOwnContext(FocusManagerImpl.java:920)
	at com.intellij.openapi.wm.impl.IdeFocusManagerImpl.runOnOwnContext(IdeFocusManagerImpl.java:130)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.actionPerformed(ActionMenuItem.java:295)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.a(ActionMenuItem.java:112)
	at com.intellij.openapi.application.TransactionGuardImpl.a(TransactionGuardImpl.java:91)
	at com.intellij.openapi.application.TransactionGuardImpl.access$300(TransactionGuardImpl.java:41)
	at com.intellij.openapi.application.TransactionGuardImpl$2.run(TransactionGuardImpl.java:116)
	at com.intellij.openapi.application.TransactionGuardImpl.submitTransaction(TransactionGuardImpl.java:126)
	at com.intellij.openapi.application.TransactionGuard.submitTransaction(TransactionGuard.java:122)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.fireActionPerformed(ActionMenuItem.java:112)
	at com.intellij.ui.plaf.beg.BegMenuItemUI.a(BegMenuItemUI.java:513)
	at com.intellij.ui.plaf.beg.BegMenuItemUI.access$300(BegMenuItemUI.java:45)
	at com.intellij.ui.plaf.beg.BegMenuItemUI$MyMouseInputHandler.mouseReleased(BegMenuItemUI.java:533)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at com.intellij.ide.IdeEventQueue.i(IdeEventQueue.java:843)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:675)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:391)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.ClassNotFoundException: com.jetbrains.cidr.cpp.cmake.projectWizard.NewCMakeProjectStepAdapter PluginClassLoader[io.github.francoiscambell.clionarduinoplugin, 1.2.2]
	at com.intellij.ide.plugins.cl.PluginClassLoader.loadClass(PluginClassLoader.java:64)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 52 more

Thanks !

schneebonus, 2017-01-09:
Same problem on **Fedora 25**

CLion 2016.**3.2**
Build #CL-163.10154.43, built on December 21, 2016
Licensed to xxxxxxxxxxxxxx
Subscription is active until xxxxxxxxxxxxx
JRE: 1.8.0_112-release-408-b6 amd64
JVM: OpenJDK 64-Bit Server VM by JetBrains s.r.o

Same Trace as everyone:
com/jetbrains/cidr/cpp/cmake/projectWizard/**NewCMakeProjectStepAdapter**
...

ltearno, 2017-01-09:
Note that it works if you install the plugin on the previous version of
CLion.

StephenStrickland, 2017-01-10:
Running CLion 3.1 on Mac OSC 10.10.4
Error creating new after clicking 'OK' button from ```File -> New Arduino Sketch Project```
stacktrace:
```
com.jetbrains.cidr.cpp.cmake.workspace.CMakeWorkspace.openProject(Lcom/intellij/openapi/vfs/VirtualFile;Lcom/intellij/openapi/project/Project;Z)Lcom/intellij/openapi/project/Project;
java.lang.NoSuchMethodError: com.jetbrains.cidr.cpp.cmake.workspace.CMakeWorkspace.openProject(Lcom/intellij/openapi/vfs/VirtualFile;Lcom/intellij/openapi/project/Project;Z)Lcom/intellij/openapi/project/Project;
	at io.github.francoiscambell.clionarduinoplugin.wizards.NewArduinoProjectWizard.doRunWizard(NewArduinoProjectWizard.java:117)
	at com.jetbrains.cidr.cpp.cmake.projectWizard.CMakeProjectWizard.runWizard(CMakeProjectWizard.java:68)
	at io.github.francoiscambell.clionarduinoplugin.actions.NewArduinoProject.actionPerformed(NewArduinoProject.java:26)
	at com.intellij.openapi.actionSystem.ex.ActionUtil$1.run(ActionUtil.java:199)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performActionDumbAware(ActionUtil.java:215)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.a(ActionMenuItem.java:315)
	at com.intellij.openapi.wm.impl.FocusManagerImpl.runOnOwnContext(FocusManagerImpl.java:920)
	at com.intellij.openapi.wm.impl.IdeFocusManagerImpl.runOnOwnContext(IdeFocusManagerImpl.java:130)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.actionPerformed(ActionMenuItem.java:295)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.a(ActionMenuItem.java:112)
	at com.intellij.openapi.application.TransactionGuardImpl.a(TransactionGuardImpl.java:91)
	at com.intellij.openapi.application.TransactionGuardImpl.access$300(TransactionGuardImpl.java:41)
	at com.intellij.openapi.application.TransactionGuardImpl$2.run(TransactionGuardImpl.java:116)
	at com.intellij.openapi.application.TransactionGuardImpl.submitTransaction(TransactionGuardImpl.java:126)
	at com.intellij.openapi.application.TransactionGuard.submitTransaction(TransactionGuard.java:122)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.fireActionPerformed(ActionMenuItem.java:112)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.JToggleButton$ToggleButtonModel.setPressed(JToggleButton.java:308)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at com.apple.laf.ScreenMenuItemCheckbox.itemStateChanged(ScreenMenuItemCheckbox.java:193)
	at java.awt.CheckboxMenuItem.processItemEvent(CheckboxMenuItem.java:389)
	at java.awt.CheckboxMenuItem.processEvent(CheckboxMenuItem.java:357)
	at java.awt.MenuComponent.dispatchEventImpl(MenuComponent.java:351)
	at java.awt.MenuComponent.dispatchEvent(MenuComponent.java:339)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:761)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at com.intellij.ide.IdeEventQueue.d(IdeEventQueue.java:843)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:679)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:391)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

rjuang, 2017-01-10:
I've been trying to update the code to reflect changes in 2016.3.2 but there seems to be a lack of documentation for CLion's cidr SDK. I was able to update the code to work with changes in 2016.3, but not 2016.3.2

In 3.2, it looks like they removed NewCMakeProjectStepAdapter but left ImportCMakeProjectStepAdapter.  I haven't figured out the equivalent for NewCMakeProjectStepAdapter. Anyone with ideas?

rjuang, 2017-01-10:
Alright, I fixed the bugs in a personal branch here: https://github.com/rjuang/CLionArduinoPlugin

vector-kerr, 2017-01-26:
@rjuang Installed your copy of the plugin and it didn't work (CLion 2016.3.2 x64, Windows). I received the following error on restart after installing the plugin. CLion version details follow the error message text below.

# Error Message
```
com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
java.lang.NoClassDefFoundError: com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
	at io.github.francoiscambell.clionarduinoplugin.actions.NewArduinoProject.actionPerformed(NewArduinoProject.java:26)
	at com.intellij.openapi.actionSystem.ex.ActionUtil$1.run(ActionUtil.java:199)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performActionDumbAware(ActionUtil.java:215)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.a(ActionMenuItem.java:315)
	at com.intellij.openapi.wm.impl.FocusManagerImpl.runOnOwnContext(FocusManagerImpl.java:920)
	at com.intellij.openapi.wm.impl.IdeFocusManagerImpl.runOnOwnContext(IdeFocusManagerImpl.java:130)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.actionPerformed(ActionMenuItem.java:295)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.a(ActionMenuItem.java:112)
	at com.intellij.openapi.application.TransactionGuardImpl.a(TransactionGuardImpl.java:91)
	at com.intellij.openapi.application.TransactionGuardImpl.access$300(TransactionGuardImpl.java:41)
	at com.intellij.openapi.application.TransactionGuardImpl$2.run(TransactionGuardImpl.java:116)
	at com.intellij.openapi.application.TransactionGuardImpl.submitTransaction(TransactionGuardImpl.java:126)
	at com.intellij.openapi.application.TransactionGuard.submitTransaction(TransactionGuard.java:122)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.fireActionPerformed(ActionMenuItem.java:112)
	at com.intellij.ui.plaf.beg.BegMenuItemUI.a(BegMenuItemUI.java:513)
	at com.intellij.ui.plaf.beg.BegMenuItemUI.access$300(BegMenuItemUI.java:45)
	at com.intellij.ui.plaf.beg.BegMenuItemUI$MyMouseInputHandler.mouseReleased(BegMenuItemUI.java:533)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at com.intellij.ide.IdeEventQueue.i(IdeEventQueue.java:843)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:675)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:391)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.ClassNotFoundException: com.jetbrains.cidr.cpp.cmake.projectWizard.NewCMakeProjectStepAdapter PluginClassLoader[io.github.francoiscambell.clionarduinoplugin, 1.2.2]
	at com.intellij.ide.plugins.cl.PluginClassLoader.loadClass(PluginClassLoader.java:64)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 52 more
```

# CLion Version Details
```
CLion 2016.3.2
Build #CL-163.10154.43, built on December 22, 2016
JRE: 1.8.0_112-release-408-b6 amd64
JVM: OpenJDK 64-Bit Server VM by JetBrains s.r.o
```

zigge, 2017-02-01:
Having the same problem, on **OSX 10.12.2**

`com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
java.lang.NoClassDefFoundError: com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
	at io.github.francoiscambell.clionarduinoplugin.wizards.NewArduinoProjectWizard.<init>(NewArduinoProjectWizard.java:21)
	at io.github.francoiscambell.clionarduinoplugin.actions.NewArduinoProject.actionPerformed(NewArduinoProject.java:26)
	at com.intellij.openapi.actionSystem.ex.ActionUtil$1.run(ActionUtil.java:199)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performActionDumbAware(ActionUtil.java:215)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.a(ActionMenuItem.java:315)
	at com.intellij.openapi.wm.impl.FocusManagerImpl.runOnOwnContext(FocusManagerImpl.java:920)
	at com.intellij.openapi.wm.impl.IdeFocusManagerImpl.runOnOwnContext(IdeFocusManagerImpl.java:130)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.actionPerformed(ActionMenuItem.java:295)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.a(ActionMenuItem.java:112)
	at com.intellij.openapi.application.TransactionGuardImpl.a(TransactionGuardImpl.java:91)
	at com.intellij.openapi.application.TransactionGuardImpl.access$300(TransactionGuardImpl.java:41)
	at com.intellij.openapi.application.TransactionGuardImpl$2.run(TransactionGuardImpl.java:116)
	at com.intellij.openapi.application.TransactionGuardImpl.submitTransaction(TransactionGuardImpl.java:126)
	at com.intellij.openapi.application.TransactionGuard.submitTransaction(TransactionGuard.java:122)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.fireActionPerformed(ActionMenuItem.java:112)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.JToggleButton$ToggleButtonModel.setPressed(JToggleButton.java:308)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at com.apple.laf.ScreenMenuItemCheckbox.itemStateChanged(ScreenMenuItemCheckbox.java:193)
	at java.awt.CheckboxMenuItem.processItemEvent(CheckboxMenuItem.java:389)
	at java.awt.CheckboxMenuItem.processEvent(CheckboxMenuItem.java:357)
	at java.awt.MenuComponent.dispatchEventImpl(MenuComponent.java:351)
	at java.awt.MenuComponent.dispatchEvent(MenuComponent.java:339)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:761)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at com.intellij.ide.IdeEventQueue.i(IdeEventQueue.java:843)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:679)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:391)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.ClassNotFoundException: com.jetbrains.cidr.cpp.cmake.projectWizard.NewCMakeProjectStepAdapter PluginClassLoader[io.github.francoiscambell.clionarduinoplugin, 1.2.2]
	at com.intellij.ide.plugins.cl.PluginClassLoader.loadClass(PluginClassLoader.java:64)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 46 more`

rjuang, 2017-02-01:
Thanks. I'll take a look this weekend. Been busy with work, so have not been able to get to any of my personal projects.

rjuang, 2017-02-02:
@vector-kerr  The error message you posted comes from the older plugin. If you examine the pull request I posted, I explicitly removed all references to NewCMakeProjectStepAdapter.

Before you install the plugin, make sure you uninstall the old one. My plugin binary is posted at:
https://github.com/rjuang/CLionArduinoPlugin/raw/master/CLionArduinoPlugin.zip

The one in francoiscampbell branch is outdated until the pull request I sent is merged.

rjuang, 2017-02-02:
@zigge The error message you posted also comes from the older plugin. The pull request I posted explicitly removes all references to NewCMakeProjectStepAdapter which shows up in your stack trace.

Before you install the plugin, make sure you uninstall the old one. My plugin binary is posted at:
https://github.com/rjuang/CLionArduinoPlugin/raw/master/CLionArduinoPlugin.zip

The one in francoiscampbell branch is outdated until the pull request I sent is merged.

krzyzanowskim, 2017-02-04:
The problem seems resolved with 1.2.3. It's just that 1.2.2 is the one available from the plugins blowser.

[image]

rjuang, 2017-02-05:
@krzyzanowskim my guess is that francoiscampbell (owner of master branch) maintains the arduino plugin version that's uploaded to jetbrains. I have no control over that.

LockonS, 2017-02-13:
Having same problems on OS X, Clion 2016.3 version. Anyone has some temporary solution to this problems?

stevieb9, 2017-02-20:
@rjuang 's external plugin works for me: https://github.com/rjuang/CLionArduinoPlugin/raw/master/CLionArduinoPlugin.zip

sergiufalcusan, 2017-02-27:
@stevieb9 how to install it manually?

elliotboney, 2017-03-14:
@rjuang I still get an exception with yours:

```
com/jetbrains/cidr/cpp/cmake/projectWizard/CLionProjectWizardUtils
java.lang.NoClassDefFoundError: com/jetbrains/cidr/cpp/cmake/projectWizard/CLionProjectWizardUtils
	at io.github.francoiscambell.clionarduinoplugin.wizards.NewArduinoProjectWizard.doRunWizard(NewArduinoProjectWizard.java:137)
	at com.jetbrains.cidr.cpp.cmake.projectWizard.CMakeProjectWizard.runWizard(CMakeProjectWizard.java:68)
	at io.github.francoiscambell.clionarduinoplugin.actions.NewArduinoProject.actionPerformed(NewArduinoProject.java:26)
	at com.intellij.openapi.actionSystem.ex.ActionUtil$1.run(ActionUtil.java:199)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performActionDumbAware(ActionUtil.java:215)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.a(ActionMenuItem.java:315)
	at com.intellij.openapi.wm.impl.FocusManagerImpl.runOnOwnContext(FocusManagerImpl.java:920)
	at com.intellij.openapi.wm.impl.IdeFocusManagerImpl.runOnOwnContext(IdeFocusManagerImpl.java:130)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.actionPerformed(ActionMenuItem.java:295)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.a(ActionMenuItem.java:112)
	at com.intellij.openapi.application.TransactionGuardImpl.a(TransactionGuardImpl.java:91)
	at com.intellij.openapi.application.TransactionGuardImpl.access$300(TransactionGuardImpl.java:41)
	at com.intellij.openapi.application.TransactionGuardImpl$2.run(TransactionGuardImpl.java:116)
	at com.intellij.openapi.application.TransactionGuardImpl.submitTransaction(TransactionGuardImpl.java:126)
	at com.intellij.openapi.application.TransactionGuard.submitTransaction(TransactionGuard.java:122)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.fireActionPerformed(ActionMenuItem.java:112)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.JToggleButton$ToggleButtonModel.setPressed(JToggleButton.java:308)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at com.apple.laf.ScreenMenuItemCheckbox.itemStateChanged(ScreenMenuItemCheckbox.java:193)
	at java.awt.CheckboxMenuItem.processItemEvent(CheckboxMenuItem.java:389)
	at java.awt.CheckboxMenuItem.processEvent(CheckboxMenuItem.java:357)
	at java.awt.MenuComponent.dispatchEventImpl(MenuComponent.java:351)
	at java.awt.MenuComponent.dispatchEvent(MenuComponent.java:339)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:761)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at com.intellij.ide.IdeEventQueue.a(IdeEventQueue.java:843)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:679)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:391)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.ClassNotFoundException: com.jetbrains.cidr.cpp.cmake.projectWizard.CLionProjectWizardUtils PluginClassLoader[io.github.francoiscambell.clionarduinoplugin, 1.2.3]
	at com.intellij.ide.plugins.cl.PluginClassLoader.loadClass(PluginClassLoader.java:64)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 47 more
```

zwmcfarland, 2017-03-16:
@rjuang 's external plugin worked for me as well. 

@sergiufalcusan To install it copy the two jars from the zip into the plugin directory for CLionArduinoPlugin, which for me was at ~/.CLion2016.3/config/plugins/CLionArduinoPlugin/lib (Linux install)

PR1YANKPAT3L, 2017-03-16:
Running CLion 2016.3.4 on Ubuntu 16.10, placing @rjuang 's plugin worked for me.

I am using JetBrains Toolbox, default location for CLion is ~/.local/share/JetBrains/Toolbox/apps/CLion/ch-0/163.13906.4/plugins/CLionArduinoPlugin

farmisen, 2017-04-12:
Running CLion 2017.1.1 on Mac OS X 10.12.3 and using @rjuang plugin worked for me - will check on Ubuntu 16.10 asap.

IamPhytan, 2017-09-22:
@zwmcfarland What is the folder for a Windows install ?

insaneshadowzaman, 2017-10-08:
Still getting the same problem.
N. B. I cannot find any folder named arduino or something similar in <dir of Clion>/plugin. Should there be any? Or the plugin is somewhere else?
Running Clion 2017.2.3 or manjaro deepin (Arch linux).
Arduino plugin version: 1.2.2
Clion location: /opt/

Error details:
`com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
java.lang.NoClassDefFoundError: com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
	at io.github.francoiscambell.clionarduinoplugin.actions.NewArduinoProject.actionPerformed(NewArduinoProject.java:26)
	at com.intellij.openapi.actionSystem.ex.ActionUtil$1.run(ActionUtil.java:216)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performActionDumbAware(ActionUtil.java:233)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.a(ActionMenuItem.java:310)
	at com.intellij.openapi.wm.impl.FocusManagerImpl.runOnOwnContext(FocusManagerImpl.java:911)
	at com.intellij.openapi.wm.impl.IdeFocusManagerImpl.runOnOwnContext(IdeFocusManagerImpl.java:136)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.actionPerformed(ActionMenuItem.java:300)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.a(ActionMenuItem.java:117)
	at com.intellij.openapi.application.TransactionGuardImpl.a(TransactionGuardImpl.java:86)
	at com.intellij.openapi.application.TransactionGuardImpl.a(TransactionGuardImpl.java:109)
	at com.intellij.openapi.application.TransactionGuardImpl.submitTransaction(TransactionGuardImpl.java:118)
	at com.intellij.openapi.application.TransactionGuard.submitTransaction(TransactionGuard.java:122)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.fireActionPerformed(ActionMenuItem.java:117)
	at com.intellij.ui.plaf.beg.BegMenuItemUI.a(BegMenuItemUI.java:513)
	at com.intellij.ui.plaf.beg.BegMenuItemUI.access$300(BegMenuItemUI.java:45)
	at com.intellij.ui.plaf.beg.BegMenuItemUI$MyMouseInputHandler.mouseReleased(BegMenuItemUI.java:533)
	at java.awt.Component.processMouseEvent(Component.java:6541)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6306)
	at java.awt.Container.processEvent(Container.java:2237)
	at java.awt.Component.dispatchEventImpl(Component.java:4897)
	at java.awt.Container.dispatchEventImpl(Container.java:2295)
	at java.awt.Component.dispatchEvent(Component.java:4719)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4889)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4526)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4467)
	at java.awt.Container.dispatchEventImpl(Container.java:2281)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4719)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:764)
	at java.awt.EventQueue.access$500(EventQueue.java:98)
	at java.awt.EventQueue$3.run(EventQueue.java:715)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:737)
	at java.awt.EventQueue$4.run(EventQueue.java:735)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:734)
	at com.intellij.ide.IdeEventQueue.i(IdeEventQueue.java:821)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:645)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:365)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.ClassNotFoundException: com.jetbrains.cidr.cpp.cmake.projectWizard.NewCMakeProjectStepAdapter PluginClassLoader[io.github.francoiscambell.clionarduinoplugin, 1.2.2] com.intellij.ide.plugins.cl.PluginClassLoader@41869cf3
	at com.intellij.ide.plugins.cl.PluginClassLoader.loadClass(PluginClassLoader.java:65)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 51 more`

A-freedom, 2018-08-14:
how can I fix it

IamPhytan, 2018-08-14:
If you are only going to use CLion with this plugin to replace Arduino IDE, consider using PlatformIO.

Found this great tutorial on YouTube : [https://www.youtube.com/playlist?list=PLyITY2JC-2OsF8vaUp17XUCeDFbg-ADzs](url)