francoiscampbell/CLionArduinoPlugin #32: Problem with CLion 2017.1 EAP

npelov, 2017-02-16:
I got this problem in CLion 2017.1 EAP. I installed plugin, restarted ide, waited for background processes to finish and selected New Arduino Sketch Project from menu.

CLion 2017.1 EAP
Build #CL-171.3224.8, built on February 14, 2017
CLion EAP User
Expiration date: March 16, 2017
JRE: 1.8.0_112-release-b702 amd64
JVM: OpenJDK 64-Bit Server VM by JetBrains s.r.o
Windows 7 6.1

<code>
com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
java.lang.NoClassDefFoundError: com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
	at io.github.francoiscambell.clionarduinoplugin.actions.NewArduinoProject.actionPerformed(NewArduinoProject.java:26)
	at com.intellij.openapi.actionSystem.ex.ActionUtil$1.run(ActionUtil.java:208)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performActionDumbAware(ActionUtil.java:224)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.a(ActionMenuItem.java:309)
	at com.intellij.openapi.wm.impl.FocusManagerImpl.runOnOwnContext(FocusManagerImpl.java:929)
	at com.intellij.openapi.wm.impl.IdeFocusManagerImpl.runOnOwnContext(IdeFocusManagerImpl.java:136)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.actionPerformed(ActionMenuItem.java:299)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.a(ActionMenuItem.java:116)
	at com.intellij.openapi.application.TransactionGuardImpl.a(TransactionGuardImpl.java:89)
	at com.intellij.openapi.application.TransactionGuardImpl.access$300(TransactionGuardImpl.java:39)
	at com.intellij.openapi.application.TransactionGuardImpl$2.run(TransactionGuardImpl.java:114)
	at com.intellij.openapi.application.TransactionGuardImpl.submitTransaction(TransactionGuardImpl.java:124)
	at com.intellij.openapi.application.TransactionGuard.submitTransaction(TransactionGuard.java:122)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.fireActionPerformed(ActionMenuItem.java:116)
	at com.intellij.ui.plaf.beg.BegMenuItemUI.a(BegMenuItemUI.java:513)
	at com.intellij.ui.plaf.beg.BegMenuItemUI.access$300(BegMenuItemUI.java:45)
	at com.intellij.ui.plaf.beg.BegMenuItemUI$MyMouseInputHandler.mouseReleased(BegMenuItemUI.java:533)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:764)
	at java.awt.EventQueue.access$500(EventQueue.java:98)
	at java.awt.EventQueue$3.run(EventQueue.java:715)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:737)
	at java.awt.EventQueue$4.run(EventQueue.java:735)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:734)
	at com.intellij.ide.IdeEventQueue.c(IdeEventQueue.java:827)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:651)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:365)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.ClassNotFoundException: com.jetbrains.cidr.cpp.cmake.projectWizard.NewCMakeProjectStepAdapter PluginClassLoader[io.github.francoiscambell.clionarduinoplugin, 1.2.2]
	at com.intellij.ide.plugins.cl.PluginClassLoader.loadClass(PluginClassLoader.java:65)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 52 more
</code>

henrycjc, 2017-02-21:
I am also receiving this error with CLION 2016.3.3 for Mac OS X El Capitan

ThYpHo0n, 2017-03-11:
Same here on Win10

Sebo-90, 2017-03-14:
Same here, also on Win10

cppgent0, 2017-03-31:
On Ubuntu 16.04.1 LTS
Just upgraded to CLion 2017.1

- worked fine in previous version of CLion
- works fine on existing Arduino CMake projects
- does not work when creating new Arduino projects

I tried this workaround:
- copy an existing working project
- change all the old names to the new names (everywhere!)

It sort of worked. I can compile, but the upload doesn't work (yet)

apps4u, 2017-04-13:
I think My is the Same issue I'm using CLion 2017.1 , Its not EAP any more but I can not get any version of pluging to work.

com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
java.lang.NoClassDefFoundError: com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
	at io.github.francoiscambell.clionarduinoplugin.wizards.NewArduinoProjectWizard.<init>(NewArduinoProjectWizard.java:21)
	at io.github.francoiscambell.clionarduinoplugin.actions.NewArduinoProject.actionPerformed(NewArduinoProject.java:26)
	at com.intellij.openapi.actionSystem.ex.ActionUtil$1.run(ActionUtil.java:215)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performActionDumbAware(ActionUtil.java:232)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.a(ActionMenuItem.java:309)
	at com.intellij.openapi.wm.impl.FocusManagerImpl.runOnOwnContext(FocusManagerImpl.java:929)
	at com.intellij.openapi.wm.impl.IdeFocusManagerImpl.runOnOwnContext(IdeFocusManagerImpl.java:136)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.actionPerformed(ActionMenuItem.java:299)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.a(ActionMenuItem.java:116)
	at com.intellij.openapi.application.TransactionGuardImpl.a(TransactionGuardImpl.java:86)
	at com.intellij.openapi.application.TransactionGuardImpl.a(TransactionGuardImpl.java:109)
	at com.intellij.openapi.application.TransactionGuardImpl.submitTransaction(TransactionGuardImpl.java:118)
	at com.intellij.openapi.application.TransactionGuard.submitTransaction(TransactionGuard.java:122)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.fireActionPerformed(ActionMenuItem.java:116)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.JToggleButton$ToggleButtonModel.setPressed(JToggleButton.java:308)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at com.apple.laf.ScreenMenuItemCheckbox.itemStateChanged(ScreenMenuItemCheckbox.java:193)
	at java.awt.CheckboxMenuItem.processItemEvent(CheckboxMenuItem.java:389)
	at java.awt.CheckboxMenuItem.processEvent(CheckboxMenuItem.java:357)
	at java.awt.MenuComponent.dispatchEventImpl(MenuComponent.java:351)
	at java.awt.MenuComponent.dispatchEvent(MenuComponent.java:339)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:767)
	at java.awt.EventQueue.access$500(EventQueue.java:98)
	at java.awt.EventQueue$3.run(EventQueue.java:715)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:737)
	at java.awt.EventQueue$4.run(EventQueue.java:735)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:734)
	at com.intellij.ide.IdeEventQueue.g(IdeEventQueue.java:827)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:655)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:365)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.ClassNotFoundException: com.jetbrains.cidr.cpp.cmake.projectWizard.NewCMakeProjectStepAdapter PluginClassLoader[io.github.francoiscambell.clionarduinoplugin, 1.2.2]
	at com.intellij.ide.plugins.cl.PluginClassLoader.loadClass(PluginClassLoader.java:65)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 45 more

gamingfail123, 2017-04-14:
Also on Ubuntu 17.04

gamingfail123, 2017-04-14:
CLion 2017.1.1

gaffo, 2017-04-15:
Same issue on ubuntu 16.10

Xorgon, 2017-05-09:
(Same issue, Windows 10).

ZipDriver, 2017-05-10:
The same problem here.

`com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
java.lang.NoClassDefFoundError: com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
	at io.github.francoiscambell.clionarduinoplugin.actions.NewArduinoProject.actionPerformed(NewArduinoProject.java:26)
	at com.intellij.openapi.actionSystem.ex.ActionUtil$1.run(ActionUtil.java:215)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performActionDumbAware(ActionUtil.java:232)
...
`
CLion 2017.1.1
Build #CL-171.4073.41, built on April 12, 2017
Licensed to CLion Evaluator
Expiration date: June 9, 2017
JRE: 1.8.0_112-release-736-b16 amd64
JVM: OpenJDK 64-Bit Server VM by JetBrains s.r.o
Windows 7 6.1

trestini, 2017-05-28:
Same here with CLion 2017.1.2 on OSX El Captain

npelov, 2017-06-17:
Creating new project works with CLion 2016.2.3 (Build #CL-162.2228.26). Then  you can open it with newer version.
One of the problems is that my arduino IDE is not installed in C:\Program Files (x86)\Arduino, but in D:\Programs\arduino-1.8.1. I had to change the path in cmake/ArduinoToolchain.cmake:

```
        set(SDK_PATH_HINTS "C:\\Program Files\\Arduino"
            "D:\\Programs\\arduino-1.8.1"
            )
```

machadolab, 2017-06-23:
Same problem :/

Danilka, 2017-07-01:
I hope this project comes back from the dead. After using JetBrains' IDE once, it's super hard to go to something like Arduino "IDE".

Having the same problem on Mac OS 2017th version.

I wish JetBrains would work on this as well. I was about to buy CLion just for this.

dmitry-cherkas, 2017-07-20:
There are forks available with this issue fixed, e.g. https://github.com/rjuang/CLionArduinoPlugin
(currently pending a pull request)

lukasMega, 2017-08-01:
the same problem :(

```
2017-08-01 21:56:17,458 [  74304]  ERROR - llij.ide.plugins.PluginManager - CLion 2017.2  Build #CL-172.3317.49 
2017-08-01 21:56:17,458 [  74304]  ERROR - llij.ide.plugins.PluginManager - JDK: 1.8.0_152-release 
2017-08-01 21:56:17,458 [  74304]  ERROR - llij.ide.plugins.PluginManager - VM: OpenJDK 64-Bit Server VM 
2017-08-01 21:56:17,458 [  74304]  ERROR - llij.ide.plugins.PluginManager - Vendor: JetBrains s.r.o 
2017-08-01 21:56:17,458 [  74304]  ERROR - llij.ide.plugins.PluginManager - OS: Windows 7 
```

```
2017-08-01 21:56:12,396 [  69242]  ERROR - llij.ide.plugins.PluginManager - com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter 
java.lang.NoClassDefFoundError: com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
	at io.github.francoiscambell.clionarduinoplugin.actions.NewArduinoProject.actionPerformed(NewArduinoProject.java:26)
	at com.intellij.ui.components.labels.ActionLink$1.linkSelected(ActionLink.java:67)
	at com.intellij.ui.components.labels.LinkLabel.doClick(LinkLabel.java:156)
	at com.intellij.ui.components.labels.ActionLink.doClick(ActionLink.java:80)
	at com.intellij.ui.components.labels.LinkLabel$MyMouseHandler.mouseReleased(LinkLabel.java:347)
	at java.awt.Component.processMouseEvent(Component.java:6541)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6306)
	at java.awt.Container.processEvent(Container.java:2237)
	at java.awt.Component.dispatchEventImpl(Component.java:4897)
	at java.awt.Container.dispatchEventImpl(Container.java:2295)
	at java.awt.Component.dispatchEvent(Component.java:4719)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4889)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4526)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4467)
	at java.awt.Container.dispatchEventImpl(Container.java:2281)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4719)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:764)
	at java.awt.EventQueue.access$500(EventQueue.java:98)
	at java.awt.EventQueue$3.run(EventQueue.java:715)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:737)
	at java.awt.EventQueue$4.run(EventQueue.java:735)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:734)
	at com.intellij.ide.IdeEventQueue.f(IdeEventQueue.java:821)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:645)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:365)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.ClassNotFoundException: com.jetbrains.cidr.cpp.cmake.projectWizard.NewCMakeProjectStepAdapter 
```