francoiscampbell/CLionArduinoPlugin #37: Doesn't work with CLion 2017.2.1 on Ubuntu Linux

volek, 2017-08-11:
Invocation of "New Arduino Scetch Project" returns the following exception:

com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
java.lang.NoClassDefFoundError: com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
	at io.github.francoiscambell.clionarduinoplugin.actions.NewArduinoProject.actionPerformed(NewArduinoProject.java:26)
	at com.intellij.openapi.actionSystem.ex.ActionUtil$1.run(ActionUtil.java:216)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performActionDumbAware(ActionUtil.java:233)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.a(ActionMenuItem.java:310)
	at com.intellij.openapi.wm.impl.FocusManagerImpl.runOnOwnContext(FocusManagerImpl.java:911)
	at com.intellij.openapi.wm.impl.IdeFocusManagerImpl.runOnOwnContext(IdeFocusManagerImpl.java:136)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.actionPerformed(ActionMenuItem.java:300)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.a(ActionMenuItem.java:117)
	at com.intellij.openapi.application.TransactionGuardImpl.a(TransactionGuardImpl.java:86)
	at com.intellij.openapi.application.TransactionGuardImpl.a(TransactionGuardImpl.java:109)
	at com.intellij.openapi.application.TransactionGuardImpl.submitTransaction(TransactionGuardImpl.java:118)
	at com.intellij.openapi.application.TransactionGuard.submitTransaction(TransactionGuard.java:122)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.fireActionPerformed(ActionMenuItem.java:117)
	at com.intellij.ui.plaf.beg.BegMenuItemUI.a(BegMenuItemUI.java:513)
	at com.intellij.ui.plaf.beg.BegMenuItemUI.access$300(BegMenuItemUI.java:45)
	at com.intellij.ui.plaf.beg.BegMenuItemUI$MyMouseInputHandler.mouseReleased(BegMenuItemUI.java:533)
	at java.awt.Component.processMouseEvent(Component.java:6541)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6306)
	at java.awt.Container.processEvent(Container.java:2237)
	at java.awt.Component.dispatchEventImpl(Component.java:4897)
	at java.awt.Container.dispatchEventImpl(Container.java:2295)
	at java.awt.Component.dispatchEvent(Component.java:4719)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4889)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4526)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4467)
	at java.awt.Container.dispatchEventImpl(Container.java:2281)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4719)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:764)
	at java.awt.EventQueue.access$500(EventQueue.java:98)
	at java.awt.EventQueue$3.run(EventQueue.java:715)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:737)
	at java.awt.EventQueue$4.run(EventQueue.java:735)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:734)
	at com.intellij.ide.IdeEventQueue.e(IdeEventQueue.java:821)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:645)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:365)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.ClassNotFoundException: com.jetbrains.cidr.cpp.cmake.projectWizard.NewCMakeProjectStepAdapter PluginClassLoader[io.github.francoiscambell.clionarduinoplugin, 1.2.2] com.intellij.ide.plugins.cl.PluginClassLoader@92a0515
	at com.intellij.ide.plugins.cl.PluginClassLoader.loadClass(PluginClassLoader.java:65)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 51 more

SystemX-Files, 2017-08-16:
Same issue here on Win 10 with CLion 2017.2.1.

Rotzbua, 2017-08-16:
This plugin do not work since 2016.3 . This would not change, because this repo is obviously abandoned (last commit ~2years).  
So this issue is just another duplicate of #36 #32 #27 #25 .

rjuang, 2017-08-30:
I started a branch here:
https://github.com/rjuang/CLionArduinoPlugin

Updated plugin that should resolve above error message:
https://github.com/rjuang/CLionArduinoPlugin/blob/master/CLionArduinoPlugin.zip

lobodol, 2017-09-12:
I've got the same error but the updated plugin provided by @rjuang looks like working.

machadolab, 2017-09-16:
Got it working using @rjuang's fork. Thanks!!

BrianHepler, 2017-10-25:
Likewise. Fixed using @rjuang fork. Thank you!

dev3011, 2017-10-27:
I tried @rjuang 's plugin with CLion 2017.2.3 on Win 7 but when I start CLion it gives me:

```
Plugin 'io.github.francoiscambell.clionarduinoplugin2017' failed to initialize and will be disabled.  Please restart CLion.

java.lang.ClassNotFoundException: io.github.francoiscambell.clionarduinoplugin.components.ArduinoFileTypeRegistration PluginClassLoader[io.github.francoiscambell.clionarduinoplugin2017, 1.2.3] com.intellij.ide.plugins.cl.PluginClassLoader@4e608658
    at com.intellij.ide.plugins.cl.PluginClassLoader.loadClass(PluginClassLoader.java:65)
    at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
    at java.lang.Class.forName0(Native Method)
    at java.lang.Class.forName(Class.java:348)
    at com.intellij.openapi.components.impl.ComponentManagerImpl.a(ComponentManagerImpl.java:380)
    at com.intellij.openapi.components.impl.ComponentManagerImpl.init(ComponentManagerImpl.java:95)
    at com.intellij.openapi.application.impl.ApplicationImpl.load(ApplicationImpl.java:421)
    at com.intellij.openapi.application.impl.ApplicationImpl.load(ApplicationImpl.java:407)
    at com.intellij.idea.IdeaApplication.run(IdeaApplication.java:203)
    at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
    at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
    at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
    at java.lang.reflect.Method.invoke(Method.java:498)
    at com.intellij.idea.MainImpl$1$1.a(MainImpl.java:64)
    at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
    at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:762)
    at java.awt.EventQueue.access$500(EventQueue.java:98)
    at java.awt.EventQueue$3.run(EventQueue.java:715)
    at java.awt.EventQueue$3.run(EventQueue.java:709)
    at java.security.AccessController.doPrivileged(Native Method)
    at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
    at java.awt.EventQueue.dispatchEvent(EventQueue.java:732)
    at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:345)
    at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
    at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
    at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
    at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
    at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
    at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```