francoiscampbell/CLionArduinoPlugin #39: Clion 2.17.3 crash at launch of New arduino project import

macfly1202, 2017-10-21:


BrianHepler, 2017-10-25:
I also have this issue. Stacktrace:
```
com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
java.lang.NoClassDefFoundError: com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
	at io.github.francoiscambell.clionarduinoplugin.actions.NewArduinoProject.actionPerformed(NewArduinoProject.java:26)
	at com.intellij.openapi.actionSystem.ex.ActionUtil$1.run(ActionUtil.java:216)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performActionDumbAware(ActionUtil.java:233)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.a(ActionMenuItem.java:310)
	at com.intellij.openapi.wm.impl.FocusManagerImpl.runOnOwnContext(FocusManagerImpl.java:911)
	at com.intellij.openapi.wm.impl.IdeFocusManagerImpl.runOnOwnContext(IdeFocusManagerImpl.java:136)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.actionPerformed(ActionMenuItem.java:300)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.a(ActionMenuItem.java:117)
	at com.intellij.openapi.application.TransactionGuardImpl.a(TransactionGuardImpl.java:86)
	at com.intellij.openapi.application.TransactionGuardImpl.a(TransactionGuardImpl.java:109)
	at com.intellij.openapi.application.TransactionGuardImpl.submitTransaction(TransactionGuardImpl.java:118)
	at com.intellij.openapi.application.TransactionGuard.submitTransaction(TransactionGuard.java:122)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.fireActionPerformed(ActionMenuItem.java:117)
	at com.intellij.ui.plaf.beg.BegMenuItemUI.a(BegMenuItemUI.java:513)
	at com.intellij.ui.plaf.beg.BegMenuItemUI.access$300(BegMenuItemUI.java:45)
	at com.intellij.ui.plaf.beg.BegMenuItemUI$MyMouseInputHandler.mouseReleased(BegMenuItemUI.java:533)
	at java.awt.Component.processMouseEvent(Component.java:6541)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6306)
	at java.awt.Container.processEvent(Container.java:2237)
	at java.awt.Component.dispatchEventImpl(Component.java:4897)
	at java.awt.Container.dispatchEventImpl(Container.java:2295)
	at java.awt.Component.dispatchEvent(Component.java:4719)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4889)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4526)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4467)
	at java.awt.Container.dispatchEventImpl(Container.java:2281)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4719)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:764)
	at java.awt.EventQueue.access$500(EventQueue.java:98)
	at java.awt.EventQueue$3.run(EventQueue.java:715)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:737)
	at java.awt.EventQueue$4.run(EventQueue.java:735)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:734)
	at com.intellij.ide.IdeEventQueue.i(IdeEventQueue.java:821)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:645)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:365)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.ClassNotFoundException: com.jetbrains.cidr.cpp.cmake.projectWizard.NewCMakeProjectStepAdapter PluginClassLoader[io.github.francoiscambell.clionarduinoplugin, 1.2.2] com.intellij.ide.plugins.cl.PluginClassLoader@1d5e4bab
	at com.intellij.ide.plugins.cl.PluginClassLoader.loadClass(PluginClassLoader.java:65)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 51 more
```

silvioprog, 2017-12-30:
Same here. Just installed and tried to use the plugin, and:

```java
com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
java.lang.NoClassDefFoundError: com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
	at io.github.francoiscambell.clionarduinoplugin.actions.NewArduinoProject.actionPerformed(NewArduinoProject.java:26)
	at com.intellij.ui.components.labels.ActionLink$1.linkSelected(ActionLink.java:67)
	at com.intellij.ui.components.labels.LinkLabel.doClick(LinkLabel.java:156)
	at com.intellij.ui.components.labels.ActionLink.doClick(ActionLink.java:80)
	at com.intellij.ui.components.labels.LinkLabel$MyMouseHandler.mouseReleased(LinkLabel.java:352)
	at java.awt.Component.processMouseEvent(Component.java:6541)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6306)
	at java.awt.Container.processEvent(Container.java:2237)
	at java.awt.Component.dispatchEventImpl(Component.java:4897)
	at java.awt.Container.dispatchEventImpl(Container.java:2295)
	at java.awt.Component.dispatchEvent(Component.java:4719)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4889)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4526)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4467)
	at java.awt.Container.dispatchEventImpl(Container.java:2281)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4719)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:764)
	at java.awt.EventQueue.access$500(EventQueue.java:98)
	at java.awt.EventQueue$3.run(EventQueue.java:715)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:737)
	at java.awt.EventQueue$4.run(EventQueue.java:735)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:734)
	at com.intellij.ide.IdeEventQueue.j(IdeEventQueue.java:822)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:646)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:366)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.ClassNotFoundException: com.jetbrains.cidr.cpp.cmake.projectWizard.NewCMakeProjectStepAdapter PluginClassLoader[io.github.francoiscambell.clionarduinoplugin, 1.2.2] com.intellij.ide.plugins.cl.PluginClassLoader@34456d9a
	at com.intellij.ide.plugins.cl.PluginClassLoader.loadClass(PluginClassLoader.java:63)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 39 more
```

stevlulz, 2018-02-19:
Sameproblem

deftkHD, 2018-06-16:
Too