francoiscampbell/CLionArduinoPlugin #40: After plugin install, "New Arduino Sketch Project" generates no response

lamothe-hub, 2018-01-23:
I just installed the plugin in through CLion. I see the option for "New Arduino Sketch Project" on the opening page, however selecting that option does nothing. No pop up window, no change of screen, nothing. 

Running 
CLion version 2016.3.2
macOS Sierra v 10.12.2

afournie, 2018-01-26:
`com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
java.lang.NoClassDefFoundError: com/jetbrains/cidr/cpp/cmake/projectWizard/NewCMakeProjectStepAdapter
	at io.github.francoiscambell.clionarduinoplugin.wizards.NewArduinoProjectWizard.<init>(NewArduinoProjectWizard.java:21)
	at io.github.francoiscambell.clionarduinoplugin.actions.NewArduinoProject.actionPerformed(NewArduinoProject.java:26)
	at com.intellij.ui.components.labels.ActionLink$1.linkSelected(ActionLink.java:67)
	at com.intellij.ui.components.labels.LinkLabel.doClick(LinkLabel.java:156)
	at com.intellij.ui.components.labels.ActionLink.doClick(ActionLink.java:80)
	at com.intellij.ui.components.labels.LinkLabel$MyMouseHandler.mouseReleased(LinkLabel.java:352)
	at java.awt.Component.processMouseEvent(Component.java:6541)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6306)
	at java.awt.Container.processEvent(Container.java:2237)
	at java.awt.Component.dispatchEventImpl(Component.java:4897)
	at java.awt.Container.dispatchEventImpl(Container.java:2295)
	at java.awt.Component.dispatchEvent(Component.java:4719)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4889)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4526)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4467)
	at java.awt.Container.dispatchEventImpl(Container.java:2281)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4719)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:764)
	at java.awt.EventQueue.access$500(EventQueue.java:98)
	at java.awt.EventQueue$3.run(EventQueue.java:715)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:737)
	at java.awt.EventQueue$4.run(EventQueue.java:735)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:734)
	at com.intellij.ide.IdeEventQueue.g(IdeEventQueue.java:822)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:646)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:366)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.ClassNotFoundException: com.jetbrains.cidr.cpp.cmake.projectWizard.NewCMakeProjectStepAdapter PluginClassLoader[io.github.francoiscambell.clionarduinoplugin, 1.2.2] com.intellij.ide.plugins.cl.PluginClassLoader@26b67f2d
	at com.intellij.ide.plugins.cl.PluginClassLoader.loadClass(PluginClassLoader.java:63)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 40 more`

reidknapp, 2018-01-30:
Same problem as @lamothe-hub however I am running 

Clion Version 2017.3.2
MacOS Sierra Version 10.12.6

MakerTim, 2018-02-21:
Same

Clion Version 2017.3.3
Windows 10 (Redstone 1)

bkuberek, 2018-03-02:
Same

Clion Version 2017.3.3
MacOS High Sierra Version 10.13.2

stevlulz, 2018-04-01:
Same
Clion Version 2017.3
Fedora WorkStation 27

nexusdk, 2018-04-13:
Same
Clion Version 2017.1.2
Gentoo with kernel 4.15.11

c4dr01d, 2018-04-22:
Same
CLion Version 2018.1.1
Windows 10 And Ubuntu 16.04.3 LTS

sklyar, 2018-05-05:
Same
CLion Version 2018.1.2
MacOS High Sierra Version 10.13.4

tarzan212, 2018-05-19:
Same problem with Clion 2018.1.2

liqihai1987, 2018-06-05:
Same 
CLion version 2018.1.3
[image]
MacOS High Sierra Version 10.13.4

BigFatLee, 2018-06-05:
1.go to https://github.com/rjuang/CLionArduinoPlugin ,and download the .zip file .
2.open CLion
3.open Configure -> plugins
4.install plugin from disk
5.choose the file .zip file in project 
6.restart CLion
7.Done

the-matti, 2018-06-07:
Same issue here.

After installing the plugin from .zip (as described by @BigFatLee) is the following error occuring:
[image]

whalemare, 2018-06-22:
Use Fork of @BigFatLee, its work! 
Thanks you!

EasyG0ing1, 2018-07-07:
@BigFatLee - can you provide a link to your fork?  Using the @rjuang fork, when I start a new Arduino project, I get this error, and I cant see any project files on the left side of the screen:

```
Couldn't read shelf information: /Users/michael/CLionProjects/SpeakerControl/CMakeLists.txt/.idea/shelf
java.nio.file.NotDirectoryException: /Users/michael/CLionProjects/SpeakerControl/CMakeLists.txt/.idea/shelf
	at sun.nio.fs.UnixFileSystemProvider.newDirectoryStream(UnixFileSystemProvider.java:406)
	at java.nio.file.Files.newDirectoryStream(Files.java:589)
	at com.intellij.configurationStore.SchemeManagerImpl.loadSchemes(SchemeManagerImpl.kt:869)
	at com.intellij.openapi.vcs.changes.shelf.ShelveChangesManager.projectOpened(ShelveChangesManager.java:169)
	at com.intellij.openapi.project.impl.ProjectManagerImpl.d(ProjectManagerImpl.java:771)
	at com.intellij.openapi.project.impl.ProjectManagerImpl.c(ProjectManagerImpl.java:387)
	at com.intellij.openapi.application.TransactionGuardImpl.a(TransactionGuardImpl.java:165)
	at com.intellij.openapi.application.TransactionGuardImpl.a(TransactionGuardImpl.java:88)
	at com.intellij.openapi.application.TransactionGuardImpl.a(TransactionGuardImpl.java:111)
	at com.intellij.openapi.application.impl.LaterInvocator$FlushQueue.a(LaterInvocator.java:447)
	at com.intellij.openapi.application.impl.LaterInvocator$FlushQueue.a(LaterInvocator.java:431)
	at com.intellij.openapi.application.impl.LaterInvocator$FlushQueue.run(LaterInvocator.java:415)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:762)
	at java.awt.EventQueue.access$500(EventQueue.java:98)
	at java.awt.EventQueue$3.run(EventQueue.java:715)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:732)
	at com.intellij.ide.IdeEventQueue.n(IdeEventQueue.java:786)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:727)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:395)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:109)
	at java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:190)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
	at java.awt.Dialog.show(Dialog.java:1077)
	at com.intellij.openapi.ui.impl.DialogWrapperPeerImpl$MyDialog.show(DialogWrapperPeerImpl.java:694)
	at com.intellij.openapi.ui.impl.DialogWrapperPeerImpl.show(DialogWrapperPeerImpl.java:426)
	at com.intellij.openapi.ui.DialogWrapper.invokeShow(DialogWrapper.java:1688)
	at com.intellij.openapi.ui.DialogWrapper.show(DialogWrapper.java:1637)
	at com.intellij.openapi.progress.util.ProgressDialog.show(ProgressDialog.java:300)
	at com.intellij.openapi.progress.util.ProgressWindow.showDialog(ProgressWindow.java:240)
	at com.intellij.openapi.progress.util.ProgressWindow.c(ProgressWindow.java:171)
	at com.intellij.openapi.application.TransactionGuardImpl$2.run(TransactionGuardImpl.java:315)
	at com.intellij.openapi.application.impl.LaterInvocator$FlushQueue.a(LaterInvocator.java:447)
	at com.intellij.openapi.application.impl.LaterInvocator$FlushQueue.a(LaterInvocator.java:431)
	at com.intellij.openapi.application.impl.LaterInvocator$FlushQueue.run(LaterInvocator.java:415)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:762)
	at java.awt.EventQueue.access$500(EventQueue.java:98)
	at java.awt.EventQueue$3.run(EventQueue.java:715)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:732)
	at com.intellij.ide.IdeEventQueue.n(IdeEventQueue.java:786)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:727)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:395)
	at com.intellij.ide.IdeEventQueue.pumpEventsForHierarchy(IdeEventQueue.java:869)
	at com.intellij.openapi.progress.util.ProgressWindow.startBlocking(ProgressWindow.java:215)
	at com.intellij.openapi.progress.util.ProgressWindow.startBlocking(ProgressWindow.java:201)
	at com.intellij.openapi.application.impl.ApplicationImpl.runProcessWithProgressSynchronously(ApplicationImpl.java:579)
	at com.intellij.openapi.progress.impl.CoreProgressManager.runProcessWithProgressSynchronously(CoreProgressManager.java:446)
	at com.intellij.openapi.progress.impl.ProgressManagerImpl.runProcessWithProgressSynchronously(ProgressManagerImpl.java:109)
	at com.intellij.openapi.progress.impl.CoreProgressManager.a(CoreProgressManager.java:318)
	at com.intellij.openapi.progress.impl.CoreProgressManager.run(CoreProgressManager.java:303)
	at com.intellij.openapi.progress.ProgressManager.run(ProgressManager.java:192)
[... cut ...]
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:395)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

whalemare, 2018-07-12:
@EasyG0ing1 https://github.com/rjuang/CLionArduinoPlugin worked for me. But any way, i dont use it, because **platformio** do it much better. Look it

EasyG0ing1, 2018-07-21:
@whalemare Do you have a link to a decent tutorial on installing and configuring platformio?

whalemare, 2018-07-25:
@EasyG0ing1 https://github.com/platformio/platform-atmelavr/tree/develop/examples/arduino-blink

EasyG0ing1, 2018-08-04:
@whalemare Thank you, I'll check it out ... I tried to get it going last week but got a little tripped up with compiling and uploading ... things just weren't flowing ... ill check out that project.  Thanks!