francoiscampbell/CLionArduinoPlugin #45: A new plugin for those interested in CLion Arduino development

vsch, 2018-11-26:
If you are interested in using CLion for Arduino development, I forked and rewrote the plugin for my use and released it as a new plugin. [JetBrains Plugin Page](https://plugins.jetbrains.com/plugin/11301-arduino-support), [Project GitHub Repo](https://github.com/vsch/CLionArduinoPlugin)

It adds Arduino and sketch and library to IDE's new project wizard instead of separate action, adds a serial port monitor tool window with auto disconnect on build option, parses `boards.txt` and `programmers.txt` to show options in dropdown lists instead of having to search for the ids in these files and edit `CMakeLists.txt` after creation.

It works on CLion 2018 so probably addresses issues #40, #44, #43

Install and create Arduino CMake projects in one click with new project wizard types and view communications with the integrated serial monitor tool window.

* `Arduino Sketch` and `Arduino Library` project types to new project wizard, with Options to select board type, cpu, programmer and port

  [image]

* `Arduino Sketch` new file action

* Serial port monitor tool window with options to disconnect on build start and reconnect on build complete.

  [image]

  [image]

  :warning: Build events are only available in CLion 2018.3 (or later) so these options are disabled if you are running an earlier version of CLion.

* Change current build parameters the `Tools` > `Arduino Support` > `Change build settings`. Preview changes to `CMakeLists.txt` before committing to them with the `Show Difference` button or use it to copy the desired changes to apply them manually.

  [image]

  [image]

* Plugin settings (Preferences on Mac)

  [image]

  * Customize template files used for creating new projects

  * Use bundled `boards.txt` and `programmers.txt` or your own versions.

  The `Show Difference` button is only displayed when a valid file is selected to let you quickly see if there are differences from the bundled version.

  The `Create From Bundled` button is only displayed when a non-existent directory is given and it can be created. Pressing it will create the directory and populate it with the bundled templates. For layout and caveats of this directory please refer to the [README.md](https://github.com/vsch/CLionArduinoPlugin/blob/master/resources/com/vladsch/clionarduinoplugin/templates/README.md) file in that directory.

#### :warning: `Serial Port Monitor` plugin cannot be used with `Arduino Support`

Both plugins use [jSSC-2.8.0] serial library and only one plugin can load the native libraries. Please uninstall or disable `Serial Port Monitor` plugin before installing `Arduino Support`.

[jSSC-2.8.0]: https://github.com/scream3r/java-simple-serial-connector

jeremykittel, 2018-12-18:
How does one include an external library for instance, I would like to include all custom libs located in '/Users/jeremykittel/Documents/Arduino/libraries'

 '/Users/jeremykittel/Documents/Arduino/libraries/Newliquidcrystal_1.3.5/'

Please advise

vsch, 2018-12-18:
@jeremykittel, you need to add to the CMakeLists.txt a line for link directories:

     link_directories(path)

Where path is the parent directory holding your libraries. One caveat is that Arduino CMake recognized libraries when the directory name (xxxx for example) they are in matches the xxxx.cpp and xxxx.h contained inside.

So your `Newliquidcrystal_1.3.5` library should have `Newliquidcrystal_1.3.5.h` and `Newliquidcrystal_1.3.5.cpp` files in it, otherwise it will not be recognized as a library.

After adding or removing an `#include` for a library you need to reload CMakeLists.txt in CLion so dependencies are rebuilt.

This was discussed in https://github.com/vsch/CLionArduinoPlugin/issues/2